pothosware/PothosFlow #108: Displaying bandpass filter output in periodogram?

TheJimD, 2016-04-02:
Is there a way to only display the positive frequencies in a periodogram?  My signal chain is:
  - SDR Source (sample rate = 2.016 MHz)
  - Freq Xlating FIR Filter (raised cosine; decimation =42; output sample rate = 48 kHz)
  - Decimating FIR Filter (CCC; raised cosine; decimation = 6; output sample rate = 8 kHz)
  - Complex to Real
  - Periodogram

This is all working fine and doing what I want except that the x-axis of the periodogram ranges from -4 to +4 kHz.  All I really want to see is 0 to 4 kHz but I can't figure out how to set the x-axis range.

Thanks for any suggestions,
    Jim D

guruofquality, 2016-04-03:
Thanks for the report. You can select a portion of the spectrum to display with a mouse. But there is no option to just permanently display the positive part of the spectrum. I'm adding an issue, it shouldn’t be much trouble: https://github.com/pothosware/pothos-plotters/issues/11

guruofquality, 2016-04-14:
Quick update, I added this feature to the plotters. They automatically detect the data type and only show the positive part of the spectrum for real-valued signals. There is also an additional manual option to always show the full spectrum or only the positive. I will probably have this feature for the next build.

closed 2016-04-14 by guruofquality