pothosware/PothosFlow #109: Desktop install - support smaller icons

guruofquality, 2016-05-11:
Some window managers handle the icon scaling, some may use the only icon size that can be found. Currently we only install 128 px icons, but 48 px is the default. Expand this script and create 48 px icons: https://github.com/pothosware/pothos-gui/blob/master/Desktop/install

sijskes, 2016-05-12:
ubuntu mate 16.04 displays normal size icons on desktop.

guruofquality, 2016-05-12:
Thanks for testing. Its probably for the best to have all of the icon sizes since some apps will not scale stuff as needed. I ended up making a convert script to generate the icons sizes and save them in the source tree. That way we don't need the convert utility at install time.

* Commit: https://github.com/pothosware/pothos-gui/commit/e17b6ad1485568523bfe761d1ad626acc834c438

closed 2016-05-12 by guruofquality