pothosware/PothosFlow #116: Plugin Module Error on OS X 10.11.5

mehdideveloper, 2016-07-10:
Hi
I installed pothos, pothosaudio, pothosblocks, pothoscomms, pothosgui, pothosplottrs, pothospython, pothossdr, pothoswidgets, soapysdr and soapyrtlsdr on my MacBook.
When I run "PothosGui", it gives lots of plugin load errors. One example:
`2016-07-10 19:33:47 Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(/usr/local/lib/Pothos/modules/blocks/libEventBlocks.so): failed safe load`

When I add a SDR Source block and set the parameters and a Spectrogram block, it shows lots of ProxyBlockEval errors. One example:

`2016-07-10 19:39:39 PothosGui.BlockEval.guiEval: Spectrogram0-Proxy Exception Message: Exception: ProxyBlockEval factory(/plotters/spectrogram)
 -> {Proxy Exception Message: Plugin Registry Error: Pothos::PluginRegistry::get(/blocks/comms/wave_trigger): plugin path not found}` 

Any help?

guruofquality, 2016-07-10:
If seemingly random modules arent loading it might be a sign of a missing runtime library. What happens when you run ``PothosUtil --load-module /usr/local/lib/Pothos/modules/blocks/libEventBlocks.so``? Hopefully there is a useful error about a missing library or something else.

mehdideveloper, 2016-07-10:
Thank you. Running the above-mentioned command gave this error:

> Plugin Module Error: Pothos::PluginModule(/usr/local/lib/Pothos/modules/blocks/libEventBlocks.so): Cannot load library: dlopen(/usr/local/lib/Pothos/modules/blocks/libEventBlocks.so, 9): Library not loaded: /usr/local/opt/poco/lib/libPocoUtil.31.dylib

I looked at /usr/local/opt/poco/lib. I have files there, but ending with "*.43.dylib", not "*.31.dylib", so I renamed every file with "43.dylib" to "31.dylib" extension. I know that this is not the right thing to do, I just did it for test.
Now PothosGui runs without any errors and I see many new blocks inside the app.
But, I still had a problem: when I added a "SDR Source" block, it did not have an input or output port and showed this error:

> SDRSource0::eval(...) - Exception: ProxyBlockEval factory(/sdr/source) -> {Exception: SDRBlock::make(): Failed ABI check. Pothos SDR 0.5-1. SoapySDR 0.5-2. Rebuild the module}

So I ran "brew reinstall pothossdr" and now all those errors are gone.
But still there should be some version mismatch or missing components because I can't run "simple_fm_demod" from "pothos-demos" repository. It says:

> PothosGui.TopologyEval.commit: Framework Topology Connect Error: Pothos::Topology::commit(): Framework Topology Connect Error: SDRSource0.activate(): Exception: RtAudio init error 'RtApiCore::probeDeviceOpen: system error (kAudioDeviceUnsupportedFormatError) setting sample rate for device (0).

Even when I do a very simple thing (connecting a "rtlsdr" SDR Source to a Spectrogram), it finds the device (says "::1: Found Rafael Micro R820T tuner") but does not show any Spectrogram without giving any errors. The last lines of log:

> 2016-07-11 01:26:00 ::1: Found Rafael Micro R820T tuner
> 2016-07-11 01:26:00 ::1: Found Rafael Micro R820T tuner
> 2016-07-11 01:26:05 SoapySDR: Using format CF32.
> 2016-07-11 01:26:05 ::1: Exact sample rate is: 1000000.026491 Hz

Thanks for your help

mehdideveloper, 2016-07-10:
Update: Sometime when I connect the RTL dongle to Macbook and then run the PothosGui, it shows this error:

> ::1: usb_claim_interface error -3

Thought maybe it's related to the above error.

guruofquality, 2016-07-10:
> I looked at /usr/local/opt/poco/lib. I have files there, but ending with ".43.dylib", not ".31.dylib", so I renamed every file with "43.dylib" to "31.dylib" extension. I know that this is not the right thing to do, I just did it for test.

Not sure whats going on with poco. It looks a lot like this issue: https://github.com/pothosware/homebrew-pothos/issues/2 Its best to rebuild pothos and toolkits against whatever poco is currently installed. Renaming the libraries like that could be very crashy

mehdideveloper, 2016-07-10:
Yes, running 'SoapySDRUtil --probe="driver=rtlsdr" ' gives success and everything's fine (I also can use the dongle in other Soapy based apps like CubicSDR)
I uninstalled pothos-python, pothos-gui and pothos and I'm trying to build them all from source.
A little bit frustrated.

mehdideveloper, 2016-07-11:
FYI, I reinstalled everything using brew and my original errors (poco related) are gone.
Now I have problems with running topologies and I've created a new issue for it

closed 2016-07-11 by guruofquality