pothosware/PothosFlow #119: Can't activate any topology

mehdideveloper, 2016-07-11:
Hi
I have installed everything using brew on Mac.
I'm trying to use an RTL-SDR and I get errors. First program: a SDR source connected to a Spectrogram.
Here's the error log:

> 2016-07-11 10:32:44 PothosGui.TopologyEval.commit: Framework Topology Connect Error: Pothos::Topology::commit(): Framework Topology Connect Error: SDRSource0.activate(): Exception: RTL-SDR device not found.
2016-07-11 10:32:45 ::1: Found Rafael Micro R820T tuner
2016-07-11 10:32:46 ::1: Found Rafael Micro R820T tuner

When I press the "Activate topology" again, it shows:

> 2016-07-11 10:41:00 SoapySDR: Using format CF32.
2016-07-11 10:41:00 ::1: Exact sample rate is: 1000000.026491 Hz

And then after some time it shows this:

> 2016-07-11 10:41:21 PothosGui.EvalEngine.monitor: Detected evaluation thread lock-up. The evaluator will not function.

What could be wrong? (FYI, I can use the RTL-SDR with other Soapy based apps like CubicSDR, so all its drivers are OK)

mehdideveloper, 2016-07-11:
Update:
Running "PothosUtil --self-tests" returns 9 failed tests (out of 120)

> Failed 9 out of 120 tests
  FAIL: /proxy/python/tests/test_basic_types
  FAIL: /proxy/python/tests/test_compare_to
  FAIL: /proxy/python/tests/test_containers
  FAIL: /proxy/python/tests/test_call_module
  FAIL: /proxy/python/tests/test_serialization
  FAIL: /proxy/python/tests/python_module_import
  FAIL: /proxy/python/tests/test_python_module
  FAIL: /proxy/python/tests/test_python_block
  FAIL: /proxy/python/tests/test_signals_and_slots

They are all the same:

>  |   Proxy Environment Factory Error: Pothos::ProxyEnvironment::make(python)                                                                                                                                                   |
 |  -> {PythonProxyEnvironment::findProxy(Pothos): dlopen(/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-dynload/_io.so, 2): Symbol not found: __PyCodecInfo_GetIncrementalDecoder |
 |   Referenced from: /usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-dynload/_io.so                                                                                                |
 |   Expected in: flat namespace

guruofquality, 2016-07-11:
I really dont know whats going on. Especially in regards to the RTL errors. It looks like the device enumeration is failing, but then it finds it after failing. I am super concerned about the dll hell thats going on here. But it looks like the application is otherwise running so maybe that is resolved..

Can you run this simple topology for rtl. No gui, i just want to see what happens:

```PothosUtil --run-topology sdr_test.json```

And this is the JSON:

```
{
    "blocks" :
    [
        {
            "id" : "Source",
            "path" : "/sdr/source",
            "args" : ["complex_float32", [0]],
            "calls" : [
                ["setupDevice", {"driver":"rtlsdr"}],
                ["setupStream", {}],
                ["setFrontendMap", ""],
                ["setClockRate", 0.0],
                ["setSampleRate", 1e6]
            ]
        },
        {
            "id" : "BlackHole",
            "path" : "/blocks/black_hole",
            "args" : []
        }
    ],
    "connections" :
    [
        ["Source", 0, "BlackHole", 0]
    ]
}

```

It should look like this:

```
PothosUtil --run-topology sdr_test.json
>>> Create Topology: sdr_test.json
Detached kernel driver
Found Rafael Micro R820T tuner
Reattached kernel driver
Detached kernel driver
Found Rafael Micro R820T tuner
2016-07-11 13:47:04 SoapySDR: Using format CF32.
Exact sample rate is: 1000000.026491 Hz
>>> Running topology, press CTRL+C to exit

```

I was typing about about the python, but the reply is here now: https://github.com/pothosware/pothos/issues/95

mehdideveloper, 2016-07-11:
I ran the above command and here's the output:

> >>> Create Topology: sdr_test.json
>>> Running topology, press CTRL+C to exit
Found Rafael Micro R820T tuner
Found Rafael Micro R820T tuner
Framework Topology Connect Error: Pothos::Topology::commit(): Framework Topology Connect Error: /sdr/source.activate(): Exception: SDRSource::activate(): device not ready

2016-07-12 01:58:32 Application: Exception

That's very strange because the device is actually ready and other apps (like CubicSDR) can find and access it.
I still haven't managed to run a topology and see any spectrums/plots.
I also can't run pothos-demos samples (like the FM demod)
Your feedback is really appreciated, I've fixed ,lots of my earlier problems.

guruofquality, 2016-07-11:
So the json topology worked just fine. I actually expected the error "device not ready". It was a known issue for scripting/deploying that was fixed. If you ``brew install pothossdr --HEAD`` it will go away. But anyway...

But this error isnt happening in the command line and it looks totally weird.
```
Exception: RTL-SDR device not found.
2016-07-11 10:32:45 ::1: Found Rafael Micro R820T tuner
2016-07-11 10:32:46 ::1: Found Rafael Micro R820T tuner
```

The first message means that the SoapyRTLSDR::SoapyRTLSDR() constructor was invoked, but it never actually enumerated any devices. But the prints after indicate that it did. I'm curious, but what did you pass into the device args, was it ``{"driver":"rtlsdr"}``?

mehdideveloper, 2016-07-12:
OK, I did "brew reinstall pothossdr --HEAD" and now the test topology runs fine, but still I've got my original problems in the PothosGui.
I'm using {"driver":"rtlsdr"} as the device args.

When I open my sample "pth" file (just a SDR source and a Spectrogram), these are shown in the log:

> [07:09:54.187000] ::1: Found Rafael Micro R820T tuner
[07:09:54.647000] ::1: Found Rafael Micro R820T tuner
[07:09:54.816000] SoapySDR: Using format CF32.
[07:09:54.823000] ::1: Exact sample rate is: 1000000.026491 Hz

Now, when I click on "Activate topology", nothing is shown. No error, no spectrum, nothing!
After two minutes or so, this error is shown:

> [07:12:19.369226] PothosGui.EvalEngine.monitor: Detected evaluation thread lock-up. The evaluator will not function.

Also, when I open "simple_fm_demod" demo, here's the output:

> [07:18:46.321000] AudioSink: Using Built-in Output through Core Audio
[07:18:46.351000] ::1: 2016-07-12 11:48:46.351 PothosUtil[17392:1714482] 11:48:46.351 WARNING: 140: This application, or a library it uses, is using the deprecated Carbon Component Manager for hosting Audio Units. Support for this will be removed in a future release. Also, this makes the host incompatible with version 3 audio units. Please transition to the API's in AudioComponent.h.
[07:18:47.073000] ::1: Found Rafael Micro R820T tuner
[07:18:47.283000] SoapySDR: Using format CF32.

Now I click on "Activate topology" and here's the errors:

> [07:19:11.194485] PothosGui.TopologyEval.commit: 
> Framework Topology Connect Error: Pothos::Topology::commit(): Framework Topology Connect Error: SDRSource0.activate(): Exception: RtAudio init error 'RtApiCore::probeDeviceOpen: system error (kAudioDeviceUnsupportedFormatError) setting sample rate for device (0).
> [07:19:12.937150] PothosGui.EnvironmentEval.update: I/O error: RemoteProxyEnvironment::transact(): connection inactive - Remote environment crashed
> [07:19:13.376000] AudioSink: Using Built-in Output through Core Audio
> [07:19:13.413000] ::1: 2016-07-12 11:49:13.413 PothosUtil[17394:1715956] 11:49:13.412 WARNING: 140: This application, or a library it uses, is using the deprecated Carbon Component Manager for hosting Audio Units. Support for this will be removed in a future release. Also, this makes the host incompatible with version 3 audio units. Please transition to the API's in AudioComponent.h.
> [07:19:14.003000] AudioSink: Using Built-in Output through Core Audio
> [07:19:14.181000] ::1: Found Rafael Micro R820T tuner
> [07:19:14.584191] PothosGui.EnvironmentEval.update: I/O error: RemoteProxyEnvironment::transact(): connection inactive - Remote environment crashed
> [07:19:15.042000] AudioSink: Using Built-in Output through Core Audio
> [07:19:15.071000] ::1: 2016-07-12 11:49:15.071 PothosUtil[17395:1716387] 11:49:15.071 WARNING: 140: This application, or a library it uses, is using the deprecated Carbon Component Manager for hosting Audio Units. Support for this will be removed in a future release. Also, this makes the host incompatible with version 3 audio units. Please transition to the API's in AudioComponent.h.
> [07:19:15.812000] ::1: Found Rafael Micro R820T tuner
> [07:19:16.013000] SoapySDR: Using format CF32.

guruofquality, 2016-07-12:
> When I open my sample "pth" file (just a SDR source and a Spectrogram), these are shown in the log:
    [07:09:54.187000] ::1: Found Rafael Micro R820T tuner
    [07:09:54.647000] ::1: Found Rafael Micro R820T tuner
    [07:09:54.816000] SoapySDR: Using format CF32.
    [07:09:54.823000] ::1: Exact sample rate is: 1000000.026491 Hz

That looks pretty normal.

> Now, when I click on "Activate topology", nothing is shown. No error, no spectrum, nothing!
After two minutes or so, this error is shown:
[07:12:19.369226] PothosGui.EvalEngine.monitor: Detected evaluation thread lock-up. The evaluator will not function.

If a call locks up, that error will happen in about 10 seconds. Whatever is going horribly wrong eventually causes a lockup, but not immediate. Interesting...

I just feel like there is an ABI library mismatch in here somewhere that would explain. I really have no idea exactly. Maybe a few ideas to try:

* You mentioned CubicSDR, just curious was that also though brew or from an installer? I'm wondering if we ended up with two versions of the same drivers both installed, but compiled with different flags or something that would cause crashy behaviour.
* If you go to the Execute menu and select "topology stats dump". There is a count for every block. It would be interesting to see what the sdr source is doing.
* Put prints in SoapyRTLSDR acquireReadBuffer(). is it being called? Whats it returning? https://github.com/pothosware/SoapyRTLSDR/blob/master/Streaming.cpp#L418

> Also, when I open "simple_fm_demod" demo, here's the output:
Now I click on "Activate topology" and here's the errors:
    [07:19:11.194485] PothosGui.TopologyEval.commit:
    Framework Topology Connect Error: Pothos::Topology::commit(): Framework Topology Connect Error: SDRSource0.activate(): Exception: RtAudio init error 'RtApiCore::probeDeviceOpen: system error (kAudioDeviceUnsupportedFormatError) setting sample rate for device (0).

It looks like the SDR source is picking up on SoapyAudio (thats where the rtaudio errors come from). Not to be confused with the audio source/sink blocks. If you remove soapyaudio or specify {"driver":"rtlsdr"} for the device args that error should go away. But that would probably leave the original issue...

mehdideveloper, 2016-07-12:
Thanks for your patience.
I built CubicSDR from source. 
Here are some ABI versions on my system:
output of "ls /usr/local/lib/libSoapy*" :

> /usr/local/lib/libSoapySDR.0.5-2.dylib	/usr/local/lib/libSoapySDR.dylib
/usr/local/lib/libSoapySDR.0.5.0.dylib

output of "ls /usr/local/lib/librtl*" :

> /usr/local/lib/librtlsdr.0.5.3.dylib  /usr/local/lib/librtlsdr.dylib
/usr/local/lib/librtlsdr.0.dylib      /usr/local/lib/librtlsdr.la
/usr/local/lib/librtlsdr.a

output of "ls /usr/local/opt/pothos/lib/" :

> libPothos.0.4-3.dylib  libPothos.0.4.0.dylib  libPothos.dylib  pkgconfig

I have a question: why don't you use the latest (0.5-x) version of Soapy* projects? When I use brew, older versions (mainly 0.4-2) are installed and when I build from source, 0.5-2 or 0.5-3 are used. It's really confusing.
Whenever I install/reinstall something, it just messes up other projects by installing/changing a new/different version.

Here's the output of SDRSource in stats viewer:

> {
    "blockName" : "SDRSource0",
    "numTaskCalls" : 30627,
    "numWorkCalls" : 29107,
    "outputStats" : [
        {
            "dtypeMarkup" : "complex_float32",
            "dtypeSize" : 8,
            "frontBytes" : 8192,
            "portAlias" : "0",
            "portName" : "0",
            "tokensEmpty" : false,
            "totalBuffers" : 16992,
            "totalElements" : 17399808,
            "totalLabels" : 2,
            "totalMessages" : 0
        }
    ],
    "tickRatioDen" : 1000000000,
    "tickRatioNum" : 1,
    "timeLastConsumed" : 0,
    "timeLastProduced" : 117892677631830,
    "timeLastWork" : 117892681091154,
    "timeStatsQuery" : 117892681108825,
    "totalTimePostWork" : 385756959,
    "totalTimePreWork" : 143710084,
    "totalTimeTask" : 17325816035,
    "totalTimeWork" : 16789259618
}

I put some prints inside "acquireReadBuffer" . Almost all of the calls fail with timeout.

> //wait for a buffer to become available
>     while (_buf_count == 0)
>     {
>         _buf_cond.wait_for(lock, std::chrono::microseconds(timeoutUs));
>         if (_buf_count == 0)
>             {
>                 printf("acquireReadBuffer: timeout\n");
>                 return SOAPY_SDR_TIMEOUT;
>             }
>     }

And sometimes it returns 8192

I see no errors now in PothosGui and all the 120 tests of PothosUtil's self-test are returned with success.
So why I don't see a plot/spectrum on the screen? Should I do anything special for the spectrogram to be shown? Do you have simple demos/examples for PothosGui, so I can run and test them? (maybe one without a device and one using a real device)

mehdideveloper, 2016-07-12:
OK Josh, I finally managed to run my own simple topology and see a real-time spectrum! (I have no idea what I've done after lots of install/reinstall/compiles, but it seems there's no problem with detecting the RTLSDR and using it inside PothosGui)

There are some glitches. For example, when I hover the mouse pointer over a graphical widget (like a spectrogram), a trail of white rectangles are created and then gone.

By the way, I haven't managed to run the "simple_fm_demod" successfully. Here's the output when I try to activate a topology:

> [13:05:33.839000] ::1: Found Rafael Micro R820T tuner
[13:05:36.258675] PothosGui.EnvironmentEval.update: I/O error: RemoteProxyEnvironment::transact(): connection inactive - Remote environment crashed
[13:05:36.744000] AudioSink: Using Built-in Output through Core Audio
[13:05:36.776000] ::1: 2016-07-12 17:35:36.776 PothosUtil[33631:1979603] 17:35:36.775 WARNING: 140: This application, or a library it uses, is using the deprecated Carbon Component Manager for hosting Audio Units. Support for this will be removed in a future release. Also, this makes the host incompatible with version 3 audio units. Please transition to the API's in AudioComponent.h.
[13:05:37.539000] ::1: Found Rafael Micro R820T tuner
[13:05:39.759000] SoapySDR: Using format CF32.

I thought maybe this demo is using older Pothos components and that's why it does not run.

guruofquality, 2016-07-12:
>  I have a question: why don't you use the latest (0.5-x) version of Soapy* projects? When I use brew, older versions (mainly 0.4-2) are installed and when I build from source, 0.5-2 or 0.5-3 are used. It's really confusing.
Whenever I install/reinstall something, it just messes up other projects by installing/changing a new/different version.

So I recently updated soapysdr to the latest tag, this time there were ABI changes and that means that the dependent projects need to be reinstalled. But the dependent projects didnt change themselves, they just need to be recompiled.

I'm not really a brew expert, and it looks as if brew does not reinstall dependent packages after dependencies are upgraded. Its a general software problem not really specific to pothos or soapysdr. Maybe I should have reinstall instructions posted.

```
brew uninstall --force pothos soapysdr
brew missing | cut -f1 -d ":" | xargs brew uninstall --force
... insert install commands from wiki here ...
```

> OK Josh, I finally managed to run my own simple topology and see a real-time spectrum! (I have no idea what I've done after lots of install/reinstall/compiles, but it seems there's no problem with detecting the RTLSDR and using it inside PothosGui)

Thats good! I assume it was a compilation nightmare that caused the problems. But its not all figured out yet

> There are some glitches. For example, when I hover the mouse pointer over a graphical widget (like a spectrogram), a trail of white rectangles are created and then gone.

Ok good to know. I didnt hear about that one before. I wonder what it could be. Are you seeing a label displaying the current frequency and power level under the mouse position? Thats what should be under the mouse at least. It does not do this on my Yosemite VM FWIW

> I thought maybe this demo is using older Pothos components and that's why it does not run.

So I saw this error and its the most concerning: ``RemoteProxyEnvironment::transact(): connection inactive - Remote environment crashed`` I have no idea whats crashing the process but it might be related to the ``deprecated Carbon Component`` warning. So thats coming from the portaudio libraries that are installed. I dont have much control over what portaudio does here.

One thing that may help here is to move blocks into the gui affinity zone. This means the blocks run in the gui process, but it also means that the gui will crash when this happens. But if you run the gui in gdb, you may find a helpful traceback: https://github.com/pothosware/pothos-gui/wiki/Tutorial#applying-an-affinity-zone

Another tip, the audio source wraps port audio, which can represent many audio devices. We dont know if its the audio source crashing, but even if so, there may be another device for the audio source that does not crash. You can put numbers in the audio source device name, 0, 1, 2, 3... to specify devices on the system. Also, the host explorer will enumerate available audio devices: https://github.com/pothosware/pothos-gui/wiki/Tutorial#the-host-explorer-panel Eventually a drop down will be automatically populated...

So thanks for all the feedback, I hope we get to the bottom of this to future proof the issue. I can already see there is a need for proper reinstall instructions for brew and python dependencies need to be fixed in case brew python is installed: https://github.com/pothosware/pothos-gui/wiki/Tutorial#the-host-explorer-panel

guruofquality, 2016-07-18:
So I think we got this issue resolved, or whatever is left is not specific to the gui or has another issue. so I'm closing. Thanks for the feedback!

* there is still an open issue about rendering which happens on some platforms: https://github.com/pothosware/pothos-gui/issues/120
* some references to the audio sink issue which I am also trying to get to the bottom of: https://groups.google.com/forum/#!topic/pothos-users/EkTBYfpBsK8

closed 2016-07-18 by guruofquality