pothosware/PothosFlow #120: Rendering problem

mehdideveloper, 2016-07-13:
There is a rendering problem in PothosGui on Mac OS X 10.11.5
In a very simple project (an SDR source connected to a Spectrogram), when I hover the mouse pointer on the Spectrogram, a trail of white rectangles is created. If I activate the topology, the rectangles disappear very fast because of the refresh rate of the Spectrogram.
I've uploaded two screenshots here: http://imgur.com/a/rcFV8

guruofquality, 2016-07-13:
@mehdideveloper Someone else saw this issue as well (Qt 5.6.0).
Just for info what version of Qt are you using? (help->about qt)

mehdideveloper, 2016-07-14:
I checked brew. I have two versions of Qt installed: 5.5.1 and 5.6.1
PothosGui is using 5.6.x as a prerequisite and it seems we can't use an earlier version.
I uninstalled Qt 5.6.1 and then tried to run PothosGui which (logically) I got an error:

> dyld: Library not loaded: /usr/local/opt/qt5/lib/QtWidgets.framework/Versions/5/QtWidgets
  Referenced from: /usr/local/bin/PothosGui
  Reason: Incompatible library version: PothosGui requires version 5.6.0 or later, but QtWidgets provides version 5.3.0

And then I tried to reinstall PothosGui by this command: brew reinstall --HEAD pothosgui
It automatically downloaded and installed Qt 5.6.1-1

So, can I use an earlier version of Qt (like 5.5)? If yes, how?

guruofquality, 2016-07-15:
@mehdideveloper Thanks for the testing, I'm not sure how to revert qt. But honestly I wouldnt. This is just something I have to replicate and fix properly. Everyone is going to use the most recent version of qt5 from homebrew or ports anyhow so it needs to work.

There is a very [similar conversation here](https://groups.google.com/forum/#!topic/pothos-users/EkTBYfpBsK8) with a [branch on pothos-gui](https://github.com/pothosware/pothos-gui/tree/force_render) that I am asking for help to test out. If you want to try as well, just make sure the --HEAD version of all pothos and pothos*.rb recipes are installed. The idea is to see if forcing a complete redraw can solve the problem, and then next step to find where it needs to be called to properly redraw after whatever is screwing it up.

guruofquality, 2016-07-20:
Update: I have two reports that this may have been fixed now. Either by adding osx bundle build mode or updating the qwt submodule. So my hopes is that this will be resolved by installing the --HEAD version of the pothos* recipes or building from master.

guruofquality, 2016-07-20:
I'm looking forward to tagging stuff and getting this out as a new release. I appreciate the help with testing!

closed 2016-07-20 by guruofquality

mehdideveloper, 2016-07-21:
A minor update:
I updated my OS X from 10.11.5 to 10.11.6 today and PothosGui's rendering went back to the previous failure state.
I reinstalled again with "--HEAD" and now everything is fine. I don't know why updating the OS impacted this.
Another issue: after reinstalling, running "PothosGui" alone returns an error:

> /usr/local/bin/PothosGui: line 3: /usr/local/bin/PothosGui.app/Contents/MacOS/PothosGui: No such file or directory

That's because it can't find the "PothosGui.app" in the same directory. One workaround is copying "PothosGui.app" from /usr/local/Cellar/pothosgui/HEAD/bin/ to /usr/local/bin

guruofquality, 2016-07-21:
Homebrew doesnt symlink the bundle, so I changed the launcher script to use the install prefix. Also I moved the .app to the install prefix and now ``brew linkapps pothosgui`` picks it up! If that ends up being weird for non-brew osx users I will have to conditionalize the bundle install path.