pothosware/PothosFlow #139: Pothos-Gui fails to execute topology, pothosutil --run-topology fmradio.json succeeds.

jerrynsr, 2017-03-11:
I'd like to run my sdr code from within the gui so I can use widgets , etc.   
for a test a opened the simple fm radio demo and modified the sdr source 0 to use my sdrplay rsp2.
If I click on the "Activate Topology" icon nothing happens.   If I export to a json file and use the pothosutil --run-topology my.json,  it works.

In the gui if I reload the plugins I get zone:[]:I/O error: remoteproxyenvironment:transact() connection inactive - Remote Environment crashed

I am running on MacOS 10.11.6
I downloaded the required dependencies and code via git and used cmake to install.

guruofquality, 2017-03-15:
Its possible that something in the rsp driver is misbehaving, possibly locking up. Since the process does not like to be nicely unloaded when there is a reload plugins. One suggestion I have is to move the SDR source block to the GUI affinity zone. See if there are either interesting prints in the terminal or any unexpected crashes. Or it might work as well... not sure.

jerrynsr, 2017-03-15:
Josh,

     Thanks! When I moved "SDR Source" to “Gui Infinity” It worked.  I no longer need to run PothosUtil to set up a proxy.  All the more reason that the folks at SDRPLAY really should open up the API,  so we can fix the driver.  I downloaded all the Pothosware demos I can find on your Wiki.  Are there going to be more ?  

I am working on converting some of the stuff I created in gnuradio.

guruofquality, 2017-03-19:
> Thanks! When I moved "SDR Source" to “Gui Infinity” It worked. I no longer need to run PothosUtil to set up a proxy. All the more reason that the folks at SDRPLAY really should open up the API, so we can fix the driver.

Glad to hear its working. I cant think of why the sdrplay would work on one server process and not another. That is worrysome, have to keep an eye on it.

> I downloaded all the Pothosware demos I can find on your Wiki. Are there going to be more ?

definitely, but I'm personally trying to prioritize framework and gui development. Certainly any volunteer efforts are appreciated. If you have any interesting demos, I would be happy to link to them or share repo access, whatever works.