pothosware/PothosFlow #140: pothos-gui can not use an affinity assigned to tcp://[::] or tcp://[::1] on macos 10.11.6 but I have a work around

jerrynsr, 2017-03-11:
I figured out a work around and will be updating my pothos-gui source.  I am not sure how to get this checked in to git but there is an issue for Mac Users.   The only way I can use my sdrplay rsp2 sdr is run PothosUtil --proxy-server=tcp://[::1]:55557 &  Configure the host in Host Explorer.  When I do that tcp://[::] , tcp://[::1], and tcp://[::1]:55557 show as connected but if I assign an affinity zone 
to tcp://[::] or tcp://[::1] it fails.   assigning an affinitty zone tp tcp://[::1]:55557 works.  simple_fm works and all the other demos work fine.   So I'll modify my source.  Maybe it's a bonjour issue ?
It works fine on linux..  (my linux vm) .  I wanted to run natively on macos to take advantage of opencl.

anyways is this what the pothosware mac developers experience ?   
My work around get's me through this for now.  Let me know if others have seen this behavior on macos 10.11.x     I have not seen this issue on linux.

jerrynsr, 2017-03-11:
[image]

guruofquality, 2017-03-15:
> When I do that tcp://[::] , tcp://[::1], and tcp://[::1]:55557 show as connected but if I assign an affinity zone to tcp://[::] or tcp://[::1] it fails. assigning an affinitty zone tp tcp://[::1]:55557 works. simple_fm works

So the GUI is also running a server on the default port, so tcp://[::] , tcp://[::1] are connecting to the server maintained by the GUI and not the one running on port 55557. `::1` is localhost, but `::` is really bind-all address, but I have seen it to also work this way in some cases.

Then `::1` should actually work in this case, but it looks like this issue you described is also the culprit here: https://github.com/pothosware/pothos-gui/issues/139

closed 2017-03-19 by guruofquality