pothosware/PothosFlow #149: Self-test failures from PothosUtil and python's role with Pothos

romeojulietthotel, 2017-05-28:
I built pothos with python3 as dependencies and the pothosgui app's working with basic FM test.  Is python3 supported? It seems to be but there're these failures.

```
Failed 11 out of 131 tests
  FAIL: /proxy/python/tests/test_basic_types
  FAIL: /proxy/python/tests/test_compare_to
  FAIL: /proxy/python/tests/test_containers
  FAIL: /proxy/python/tests/test_call_module
  FAIL: /proxy/python/tests/test_serialization
  FAIL: /proxy/python/tests/test_numpy_array
  FAIL: /proxy/python/tests/test_numpy_types
  FAIL: /proxy/python/tests/python_module_import
  FAIL: /proxy/python/tests/test_python_module
  FAIL: /proxy/python/tests/test_python_block
  FAIL: /proxy/python/tests/test_signals_and_slots
```

I looked here and see that it seems to be python related but I really don't know what is python's role here.  Does building the software need python stuff for python2 and python3 or just one or the other?

https://github.com/pothosware/pothos-gui/issues/119

guruofquality, 2017-05-28:
Both python2 and python3 are supported, but only one can actually be installed at once due to the fact that you cant link the same app against both versions of python at once (symbols clash).

You don't need python for anything unless you are making and or using blocks written in python: https://github.com/pothosware/pothos-python/wiki#python-block-api So the unit tests failing should not harm anything.

Chances are that the unit tests are failing because either PYTHONPATH is not set or python shared library is not found.  You might get more information by running `PothosUtil --self-test /proxy/python/tests/test_basic_types` and seeing if any interesting errors are printed.

Its also good to check that output of the cmake command which shows a summary of python includes and libraries. Sometimes it gets confused to find the correct library and corresponding headers between multiple versions installed on the system.

Also, you can force the build to pick a version of python using the following variables passed to cmake
```
-DPYTHON_EXECUTABLE=${PYTHON3_EXECUTABLE}
-DPYTHON_INCLUDE_DIR=${PYTHON3_INCLUDE_DIR}
-DPYTHON_LIBRARY=${PYTHON3_LIBRARY}
```

romeojulietthotel, 2017-05-29:
Okay, I was looking in the python usr/lib/python3.5/site-packages and saw _SoapySDR.so and for some reason thought I had what I needed. 

I did set PYTHON_EXECUTABLE but I also set -DPOTHOS_PYTHON_DIR.

I checked the output of the cmake and it looks right.  I think maybe something did not end up in the right place at "make install".

romeojulietthotel, 2017-05-29:
I found that I had a typo in a cmake -D variable so the "make install" was the problem but due to the typo.

closed 2017-05-29 by romeojulietthotel