pothosware/PothosFlow #152: Pothos not running on MacOS Sierra 10.12

mtossain, 2017-06-29:
Dear all,

I tried to make Pothos run on MacOS with LimeSDR, but without any success.

1) I installed everything according to the instructions on the build page using brew.

2) I can run PothosGUI, but then when I run a simple example, the plotters do not show up... nothing happens

3) I try to run a simple python example and then I get the error that SoapySDR module is not found...

4) I use Anaconda python, and it seems that the SoapySDR is not in the path or something. I tried to install with the python3 flag, but that did not help either...

Can anybody help?

guruofquality, 2017-06-29:
> the plotters do not show up

Does this help?
https://github.com/pothosware/pothos-gui/wiki/Tutorial#insert-graphical-widget

> I try to run a simple python example and then I get the error that SoapySDR module is not found...

I think it would help to know where files are installed

Also what does `SoapySDRUtil` --info show

mtossain, 2017-06-29:
[image]
I could add the spectogram  to the block diagram, but when activating nothing happens...

Here the output of SaopySDRUtil:
SoapySDRUtil --info 
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Lib Version: v0.6.0-release
API Version: v0.6.0
ABI Version: v0.6
Install root: /usr/local
Search path: /usr/local/lib/SoapySDR/modules0.6
Module found: /usr/local/lib/SoapySDR/modules0.6/libHackRFSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.6/libLMS7Support.so
Module found: /usr/local/lib/SoapySDR/modules0.6/libRedPitaya.so
Module found: /usr/local/lib/SoapySDR/modules0.6/libremoteSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.6/librtlsdrSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.6/libuhdSupport.so
Loading modules... Mac OS; Clang version 8.1.0 (clang-802.0.41); Boost_106400; UHD_003.010.000.000-0-unknown

done
Available factories...hackrf, lime, null, redpitaya, remote, rtlsdr, uhd

Output of find;

 SoapySDRUtil --find 
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Mac OS; Clang version 8.1.0 (clang-802.0.41); Boost_106400; UHD_003.010.000.000-0-unknown

Found device 0
  addr = 1d50:6108
  driver = lime
  label = LimeSDR-USB [USB 3.0] 9061C02CD1709
  media = USB 3.0
  module = STREAM
  name = LimeSDR-USB
  serial = 0009061C02CD1709

But then output of soapy power:

soapy_power -f 87.0M:108.0M -B 10.0k -T 1.0 -d -r 2560000.0 -p 0 -F soapy_power_bin --output-fd 15 -g 37.0 -c --even --fft-window boxcar --remove-dc
Traceback (most recent call last):
  File "/Applications/Anaconda3/anaconda/bin/soapy_power", line 7, in <module>
    from soapypower.__main__ import main
  File "/Applications/Anaconda3/anaconda/lib/python3.5/site-packages/soapypower/__main__.py", line 5, in <module>
    import simplesoapy
  File "/Applications/Anaconda3/anaconda/lib/python3.5/site-packages/simplesoapy.py", line 5, in <module>
    import SoapySDR
ImportError: No module named 'SoapySDR'

guruofquality, 2017-06-29:
> I could add the spectogram to the block diagram, but when activating nothing happens...

It looks the same for me. 500e3 is at nyquist when the sample rate is 1e6, so the tone isnt going to be on the very edge of the plot

as far as the python, can you show the output of brew install with verbose? `brew install soapysdr --verbose` I think. I'm curious where the modules are getting installed.

Also, can you try --with-python3? `brew install soapysdr --with-python3` https://github.com/pothosware/homebrew-pothos/wiki#python3-support

mtossain, 2017-06-29:
dear guru
managed to get the pothos working with the graph... now i need to find the proper Limesdr settings... i did not know that after adding the block, we also need to add the graph...

for the soapy, I did actually as on the mac install page, with using brew install soapysdr --with-python3
but then it appears that still 2.7 is used for the soapy library...
[image]

guruofquality, 2017-06-29:
Can you attach the output of `brew install soapysdr --with-python3 --verbose` I dont know why its not finding python3, so the build verbose should explain that.

mtossain, 2017-06-29:
brew install soapysdr --with-python3 -verbose
==> Installing soapysdr from pothosware/pothos
==> Downloading https://github.com/pothosware/SoapySDR/archive/soapy-sdr-0.6.0.tar.gz
Already downloaded: /Users/micheltossaint/Library/Caches/Homebrew/soapysdr-0.6.0.tar.gz
==> Verifying soapysdr-0.6.0.tar.gz checksum
tar xzf /Users/micheltossaint/Library/Caches/Homebrew/soapysdr-0.6.0.tar.gz
==> cmake .. -DUSE_PYTHON_CONFIG=ON -DENABLE_PYTHON3=ON -DSOAPY_SDR_EXTVER=release -DSOAPY_SDR_ROOT='/usr/local' -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/soapysdr/0.6.0 -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_VERBOSE_MAKEFILE=ON -Wno-dev
-- The C compiler identification is AppleClang 8.1.0.8020042
-- The CXX compiler identification is AppleClang 8.1.0.8020042
-- Check for working C compiler: /usr/local/Homebrew/Library/Homebrew/shims/super/clang
-- Check for working C compiler: /usr/local/Homebrew/Library/Homebrew/shims/super/clang -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/local/Homebrew/Library/Homebrew/shims/super/clang++
-- Check for working CXX compiler: /usr/local/Homebrew/Library/Homebrew/shims/super/clang++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- 
-- #############################################
-- ## Begin configuration for Python support...
-- #############################################
-- Enabling optional Python bindings if possible...
-- Found SWIG: /usr/local/bin/swig (found version "3.0.12") 
-- SWIG_FOUND: TRUE - 3.0.12
-- Found PythonInterp: /usr/local/bin/python (found version "2.7.13") 
-- PYTHONINTERP_FOUND: TRUE - 2.7.13
-- PYTHON_EXECUTABLE: /usr/local/bin/python
-- PYTHON_INSTALL_DIR: ${prefix}/lib/python2.7/site-packages
-- PYTHONLIBS_FOUND: TRUE - 2.7.13
-- PYTHON_INCLUDE_DIRS: /usr/local/Cellar/python/2.7.13/Frameworks/Python.framework/Versions/2.7/include/python2.7;/usr/local/Cellar/python/2.7.13/Frameworks/Python.framework/Versions/2.7/include/python2.7
-- PYTHON_LIBRARIES: -L/usr/local/opt/python/Frameworks/Python.framework/Versions/2.7/lib/python2.7/config -lpython2.7 -ldl -framework CoreFoundation
-- Performing Test SIZE_T_IS_UNSIGNED_INT
-- Performing Test SIZE_T_IS_UNSIGNED_INT - Failed
-- CMAKE_SWIG_FLAGS=-c++;-threads
-- 
-- #############################################
-- ## Begin configuration for Python3 support...
-- #############################################
-- Enabling optional Python3 bindings if possible...
-- SWIG_FOUND: TRUE - 3.0.12
-- Found Python3Interp: /usr/local/bin/python3  
-- Could NOT find Python3InterpDbg (missing:  PYTHON3_DBG_EXECUTABLE) 
-- PYTHON3INTERP_FOUND: TRUE
-- PYTHON3_EXECUTABLE: /usr/local/bin/python3
-- PYTHON3_INSTALL_DIR: ${prefix}/lib/python3.6/site-packages
-- Found Python3Libs: -L/usr/local/opt/python3/Frameworks/Python.framework/Versions/3.6/lib/python3.6/config-3.6m-darwin -lpython3.6m -ldl -framework CoreFoundation  
-- PYTHON3LIBS_FOUND: TRUE
-- PYTHON3_INCLUDE_DIRS: /usr/local/Cellar/python3/3.6.1/Frameworks/Python.framework/Versions/3.6/include/python3.6m;/usr/local/Cellar/python3/3.6.1/Frameworks/Python.framework/Versions/3.6/include/python3.6m
-- PYTHON3_LIBRARIES: -L/usr/local/opt/python3/Frameworks/Python.framework/Versions/3.6/lib/python3.6/config-3.6m-darwin -lpython3.6m -ldl -framework CoreFoundation
-- CMAKE_SWIG_FLAGS=-c++;-threads
-- 
-- ######################################################
-- ## SoapySDR enabled features
-- ######################################################
-- 
 * Library, runtime library v0.6.0-release
 * Tests, library unit tests
 * Python, python bindings v2.7.13
 * Python3, python3 bindings

-- ######################################################
-- ## SoapySDR disabled features
-- ######################################################
-- 
 * Docs, doxygen documentation
[... 159 lines cut ...]
-- Installing: /usr/local/Cellar/soapysdr/0.6.0/lib/libSoapySDR.0.6.dylib
-- Installing: /usr/local/Cellar/soapysdr/0.6.0/lib/libSoapySDR.dylib
-- Installing: /usr/local/Cellar/soapysdr/0.6.0/lib/pkgconfig/SoapySDR.pc
-- Installing: /usr/local/Cellar/soapysdr/0.6.0/bin/SoapySDRUtil
-- Installing: /usr/local/Cellar/soapysdr/0.6.0/share/man/man1/SoapySDRUtil.1
-- Installing: /usr/local/Cellar/soapysdr/0.6.0/lib/python2.7/site-packages/_SoapySDR.so
-- Installing: /usr/local/Cellar/soapysdr/0.6.0/lib/python2.7/site-packages/SoapySDR.py
-- Installing: /usr/local/Cellar/soapysdr/0.6.0/lib/python3.6/site-packages/_SoapySDR.so
-- Installing: /usr/local/Cellar/soapysdr/0.6.0/lib/python3.6/site-packages/SoapySDR.py
==> Cleaning
==> Finishing up
ln -s ../Cellar/soapysdr/0.6.0/bin/SoapySDRUtil SoapySDRUtil
ln -s ../Cellar/soapysdr/0.6.0/include/SoapySDR SoapySDR
ln -s ../../Cellar/soapysdr/0.6.0/share/cmake/SoapySDR SoapySDR
ln -s ../../../Cellar/soapysdr/0.6.0/share/man/man1/SoapySDRUtil.1 SoapySDRUtil.1
ln -s ../Cellar/soapysdr/0.6.0/lib/libSoapySDR.0.6.0.dylib libSoapySDR.0.6.0.dylib
ln -s ../Cellar/soapysdr/0.6.0/lib/libSoapySDR.0.6.dylib libSoapySDR.0.6.dylib
ln -s ../Cellar/soapysdr/0.6.0/lib/libSoapySDR.dylib libSoapySDR.dylib
ln -s ../../Cellar/soapysdr/0.6.0/lib/pkgconfig/SoapySDR.pc SoapySDR.pc
ln -s ../../../Cellar/soapysdr/0.6.0/lib/python2.7/site-packages/SoapySDR.py SoapySDR.py
ln -s ../../../Cellar/soapysdr/0.6.0/lib/python2.7/site-packages/_SoapySDR.so _SoapySDR.so
ln -s ../../../Cellar/soapysdr/0.6.0/lib/python3.6/site-packages/SoapySDR.py SoapySDR.py
ln -s ../../../Cellar/soapysdr/0.6.0/lib/python3.6/site-packages/_SoapySDR.so _SoapySDR.so
==> Summary
🍺  /usr/local/Cellar/soapysdr/0.6.0: 37 files, 2.4MB, built in 18 seconds

mtossain, 2017-06-29:
also tried to install soapy_power by pip, but same result:

pip install soapy_power
Collecting soapy_power
  Using cached soapy_power-1.6.1.tar.gz
Collecting numpy (from soapy_power)
  Downloading numpy-1.13.0-cp27-cp27m-macosx_10_6_intel.macosx_10_9_intel.macosx_10_9_x86_64.macosx_10_10_intel.macosx_10_10_x86_64.whl (4.6MB)
    100% |████████████████████████████████| 4.6MB 313kB/s 
Collecting simplesoapy>=1.5.0 (from soapy_power)
  Using cached SimpleSoapy-1.5.1.tar.gz
Collecting simplespectral (from soapy_power)
  Using cached SimpleSpectral-1.0.0.tar.gz
Building wheels for collected packages: soapy-power, simplesoapy, simplespectral
  Running setup.py bdist_wheel for soapy-power ... done
  Stored in directory: /Users/micheltossaint/Library/Caches/pip/wheels/e7/16/b0/fd737ddf007f51312a1792701497ae03375a9c9c7b5d941ccc
  Running setup.py bdist_wheel for simplesoapy ... done
  Stored in directory: /Users/micheltossaint/Library/Caches/pip/wheels/55/78/c0/6f04abebd1b498a6d036e3c785224b630a97a055053695e3d5
  Running setup.py bdist_wheel for simplespectral ... done
  Stored in directory: /Users/micheltossaint/Library/Caches/pip/wheels/56/44/8b/eadd4aa7d6a8dfe5f6a33ab36b80c2af3e625fc9bb10605562
Successfully built soapy-power simplesoapy simplespectral
Installing collected packages: numpy, simplesoapy, simplespectral, soapy-power
Successfully installed numpy-1.13.0 simplesoapy-1.5.1 simplespectral-1.0.0 soapy-power-1.6.1
17:58 micheltossaint@dhcp-10-176-21-201 /Users/micheltossaint
% soapy_power -f 87.0M:108.0M -B 10.0k -T 1.0 -d -r 2560000.0 -p 0 -F soapy_power_bin --output-fd 15 -g 37.0 -c --even --fft-window boxcar --remove-dc
Traceback (most recent call last):
  File "/Applications/Anaconda3/anaconda/bin/soapy_power", line 7, in <module>
    from soapypower.__main__ import main
  File "/Applications/Anaconda3/anaconda/lib/python3.5/site-packages/soapypower/__main__.py", line 5, in <module>
    import simplesoapy
  File "/Applications/Anaconda3/anaconda/lib/python3.5/site-packages/simplesoapy.py", line 5, in <module>
    import SoapySDR
ImportError: No module named 'SoapySDR'
17:58 micheltossaint@dhcp-10-176-21-201 /Users/micheltossaint
%

guruofquality, 2017-06-29:
Looks like brew used python3.6 and this Anaconda3 is using python3.5. So these are two completely different installations of python. Can you install soapypower under the homebrew python environment instead?

mtossain, 2017-06-29:
thanks so much for your help!!!
how would i install soapypower with homebrew?

guruofquality, 2017-06-29:
No idea, but you probably have a lot of options:

* is there an existing brew package for soapy power?
* make a new brew package for soapy power
* build soapy power from source against homebrew python environment
* build soapysdr from source against anaconda3 python environment (you can point soapysdr to the plugins that brew installed for hardware support with SOAPY_SDR_PLUGIN_PATH)

If you figure it out, let us know, so it can be documented, thanks!

mtossain, 2017-07-08:
dear Josh

Tried to find out a bit further and found out this when installing SoapySDR from source:
[image]

It seems the python library is build by the compile, but it is not copied to any python library directory nor to the anaconda library directory. Could it be that there is a step missing in the build?

Thanks 
Michel