pothosware/PothosFlow #158: TCP Server Sink block issue

theman00011, 2017-11-09:
When configuring the TCP Server Sink block, everything except the "Nonblocking mode" evaluates. The error under the textbox is "Exception: EvalEnvironment::eval(): expression is empty". This happens with both "On" and "Off" selected, as well as manually typing in "True" or "False". As a result, the TCP sink can't be used because it believes that property is empty. I'm running Pothos 0.5.1 on Windows 10 currently.

guruofquality, 2017-11-10:
Its probably a generation issue here: https://github.com/pothosware/gr-pothos/wiki I will update when it fixed. But you should be able to manually enter the value of true or false, no quotes if you put the parameter into lineedit mode (click the little arrow to the right of the parameter widget) and you can enter any arbitrary input.

theman00011, 2017-11-10:
Unfortunately, even in lineedit mode it still doesn't evaluate with the same error.

guruofquality, 2017-11-10:
Fixed upstream, generation wise the combobox expected a string, so it was funny for bool and integer types: https://github.com/pothosware/gr-pothos/commit/64c0ca0978ab494c7a01f805eb20644abda49387

theman00011, 2017-11-12:
Awesome, I will wait for the binary distribution to catch up because building from source on Windows 10 seems to be a huge pain.