pothosware/PothosFlow #159: System Info tree is not displaying devices correctly

ashley-b, 2017-11-17:
Hi,
I have noticed that System Info tree is displaying devices just in the root, I think this should be grouped like the other data perhaps under "Devices" -> Device name -> then the device info.

[image]

Also I have noticed that audio device does not show up in this list, I assume its because the json is a list, were soapy returns a map.
```
PothosUtil --device-info

>>> Specify --device-info=deviceType for more information...
>>> Available device types in the plugin tree are: 
  * soapy
  * audio
```
<hr>

```
PothosUtil --device-info=audio
>>> Querying device info: /devices/audio/info
ALSA lib pcm_dmix.c:1099:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm.c:2501:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2501:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2501:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm_route.c:867:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_dmix.c:1099:(snd_pcm_dmix_open) unable to open slave
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
[
    {
        "Default Sample Rate": 44100,
        "Device Name": "HDA Intel PCH: ALC887-VD Analog (hw:0,0)",
        "Host API Name": "ALSA",
        "Max Input Channels": 2,
        "Max Output Channels": 0
    },
    {
        "Default Sample Rate": 44100,
        "Device Name": "HDA Intel PCH: ALC887-VD Alt Analog (hw:0,2)",
        "Host API Name": "ALSA",
        "Max Input Channels": 2,
        "Max Output Channels": 0
    },
    {
        "Default Sample Rate": 44100,
        "Device Name": "HDA ATI HDMI: 0 (hw:1,3)",
        "Host API Name": "ALSA",
        "Max Input Channels": 0,
        "Max Output Channels": 8
    },
    {
        "Default Sample Rate": 48000,
        "Device Name": "sysdefault",
        "Host API Name": "ALSA",
        "Max Input Channels": 128,
        "Max Output Channels": 0
    },
    {
        "Default Sample Rate": 44100,
        "Device Name": "pulse",
        "Host API Name": "ALSA",
        "Max Input Channels": 32,
        "Max Output Channels": 32
    },
    {
        "Default Sample Rate": 44100,
        "Device Name": "default",
        "Host API Name": "ALSA",
        "Max Input Channels": 32,
        "Max Output Channels": 32
    }
]
```

<hr>

```
PothosUtil --device-info=soapy
>>> Querying device info: /devices/soapy/info
linux; GNU C++ version 7.1.1 20170622 (Red Hat 7.1.1-3); Boost_106300; UHD_003.010.002.HEAD-0-gbd6e21dc

{
    "SDR Device": [
        {
            "default_input": "True",
            "default_output": "True",
            "device_id": "0",
            "driver": "audio",
            "label": "PulseAudio"
        }
    ],
    "SoapySDR info": {
        "ABI Version": "0.6",
        "API Version": "0.6.0",
        "Factories": "audio, null, remote, uhd",
        "Install Root": "/usr/local"
    }
}
```

guruofquality, 2017-11-18:
FWIW, that was a recent typo in the audio blocks from the JSON rewrite. So the audio should look much like the SDR devices with the recent fix. Thanks for finding that. -- However the display was actually intended where the array of key/value objects are displayed basically up one level with the index number for each entry. The display does make sure only the first element is actually expanded so it doesn't unnecessarily take up all of the available space.

ashley-b, 2017-11-21:
Fair enough in regards to the System Info Tree layout.
Just did a rebuild with the JSON fix in Audio block, all looks good.

Thanks