pothosware/PothosFlow #162: Windows 7 Audio Source: Invalid number of channels

piotrva, 2018-03-03:
I am using Windows 7 x64.

As I try to use Audio Source block (non GnuRadio), regardless on settings I get:
`Pa_IsFormatSupported: Invalid number of channels`

I have tried different devices, sample rates, data types, num channels (from 1 to 4) as well as channel mode - with same result.

guruofquality, 2018-03-08:
Have you tried changing the selected audio device? Usually many devices will show up for port audio, and not all of them are useful.

You can see the available devices with the command line `PothosUtil --device-info=audio` or..

The host explorer panel will list the audio devices too: https://github.com/pothosware/PothosFlow/wiki/Tutorial#the-host-explorer-panel

piotrva, 2018-03-10:
1. After typing the command I see following messages:
[image]

2. The host explorer shows as follows:
[image]

3. I have tried different Device Names (typing integer numbers: 0, 1, 2, ...) - also same result for each setting - the device seems to change, but stil there is an error:
[image]
[image]

dbnut, 2018-04-06:
I have the identical problem with Win 7 (64). And the same difficulty occurs with Audio Source in Gnu Radio Companion (no block connectors, but without the benefit of explicit error messages).

Note that I am connecting to any available audio device via its index number, with the corresponding name shown in the Message Window. The same error applies to *all* devices, for all APIs (MME, WASAPI,WDM-KS).

Note also that Audio Sink appears to function correctly and can be configured for number of channels/ports, etc.

Note I also have the "failed safe load" message when adding the first block to a graph.

Thanks for an otherwise great product. Looking forward to a solution soon - there seems to be no alternative to achieve my particular goal :).

Kind regards
Paul

dbnut, 2018-04-06:
P.S. Seeing the error for a default Num channels =1 rather suggests the device properties always have Max Output Channels = 0.

guruofquality, 2018-04-08:
Found the bug https://github.com/pothosware/PothosAudio/commit/ac2f9074b973606bfc6bd3b6d6e5ca06604c55e8

It will be in the next windows build

closed 2018-04-08 by guruofquality

dbnut, 2018-04-08:
Quick work, Sir, very impressed. Thanks.

piotrva, 2018-04-15:
@guruofquality - is there a possibility to use audio output device as an Audio Source in Pothos? When I set so I get same error as mentioned in the issue.

dbnut, 2018-04-15:
@piotrva, Sorry! Just realised you posted screenshots before. But...
What do you mean by using an "audio output device" as an Audio Source?
I do find the whole "device" system confusing in Windows. If I change the Windows Default recording device to (say) a Virtual Audio Cable *and* use the corresponding device id from host device list *then* i can receive audio via Audio Source.
Unfortunately it is not clear that I can receive from non-default devices but i have not investigated this properly. One annoying thing is that selecting device ids out of range does not produce an error, and we have no confirmation at *block* level what the device "friendly" name is.
BTW, I got Audio Source working from a re-download of the Gnu Radio binary available on 5 April. Not at all sure if the latest Pothos has that bug fix.

piotrva, 2018-04-15:
@dbnut The bug is fixed in the mentioned by you release.
I mean selecting for example Microsoft Sound Mapping - Output, as presented below:
[image]

Some applications allow to capture output signal.

I know that Windows is strange in some cases (therefore I have installed second OS on my PC - the UNIX one), but for some applications i have to use this OS.

dbnut, 2018-04-15:
@piotrva 
Yes, believe that should work *if* you make it the Windows default recording device.
Sorry, but cannot check right now, and don't have any other useful experience.
It would be SUPER HELPFUL to have a POTHOS/GRC FOR DUMMIES user guide :)

dbnut, 2018-04-15:
@piotrva
BTW, I don't have Pothos Flow working at the moment, use only Gnu Radio Companion. This is *much* poorer as it does not have a drop-down with device names. That's one of the reasons my progress is so slow.
BTW2, have you tried using Virtual Audio Cable or VB Cable to make your audio routing more flexible? These do show as available audio endpoints, and can help with Windows sound architecture limitations.