pothosware/PothosFlow #170: Feature Request: Notes/Comments in flow graph

wluker, 2018-11-14:
It might be useful to have a non-executable graphical block for adding comments to PothosFlow graphs.

Hopefully I am not annoying you with all these feature requests. Please feel free to ignore them. I just write them as I stumble around while learning Pothos.

guruofquality, 2019-01-06:
> Hopefully I am not annoying you with all these feature requests. Please feel free to ignore them. I just write them as I stumble around while learning Pothos.

No its great. Having time to do things is the issue. Leave as many TODO requests and wishlist ideas as you can think of.

dbnut, 2020-01-23:
Gnu Radio has comments attached to blocks. That's good, too.