pothosware/PothosFlow #174: PothosFlow crashes when closing undocked topology tabs

ashley-b, 2019-03-04:
Closing undocked topology via "x" button on dialog will crash PothosFlow, but no always the first time.

Host info
   Qt version 5.11.3
   Fedora 29
   revision ae031da

> 

```
#0  0x0000000000827da0 in  ()
#1  0x00007ffff5436a3a in QWindow::destroy() () at /lib64/libQt5Gui.so.5
#2  0x00007ffff54375cf in QWindow::event(QEvent*) () at /lib64/libQt5Gui.so.5
#3  0x00007ffff59d8d68 in  () at /lib64/libQt5Widgets.so.5
#4  0x00007ffff597c285 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib64/libQt5Widgets.so.5
#5  0x00007ffff59839a0 in QApplication::notify(QObject*, QEvent*) () at /lib64/libQt5Widgets.so.5
#6  0x00007ffff5081ec6 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib64/libQt5Core.so.5
#7  0x00007ffff542860a in QGuiApplicationPrivate::processCloseEvent(QWindowSystemInterfacePrivate::CloseEvent*) () at /lib64/libQt5Gui.so.5
#8  0x00007ffff542d605 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) () at /lib64/libQt5Gui.so.5
#9  0x00007ffff540870b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib64/libQt5Gui.so.5
#10 0x00007ffff0f1585f in  () at /lib64/libQt5XcbQpa.so.5
#11 0x00007ffff5080e0b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib64/libQt5Core.so.5
#12 0x00007ffff5088ed6 in QCoreApplication::exec() () at /lib64/libQt5Core.so.5
#13 0x00000000004a06b1 in main(int, char**) (argc=1, argv=0x7fffffffcfa8) at pothos/flow/PothosFlow.cpp:84
```

guruofquality, 2019-03-05:
I'm not seeing it ubuntu 18.04, so I will have to spin up a VM to replicate it. Maybe a Qt bug or particular to the version of Qt. Hmmm

ashley-b, 2019-03-05:
Yeah its an odd one, its only some times.
If i use the escape key to close the dialog I can not get it to segfault, only with mouse.
I put some debug lines in to see what's happening
"events:" are printed from DockingTabWidget::eventFilter and handleDialogFinished is printed from slot.

2019-03-05 22:31:13 default: events:  "Paint" ,  ""
2019-03-05 22:31:13 default: events:  "Close" ,  ""
2019-03-05 22:31:13 default: events:  "WindowDeactivate" ,  ""
2019-03-05 22:31:13 default: events:  "ActivationChange" ,  ""
2019-03-05 22:31:13 default: events:  "Hide" ,  ""
2019-03-05 22:31:13 default: events:  "HideToParent" ,  ""
2019-03-05 22:31:13 default: handleDialogFinished
2019-03-05 22:31:13 default: events:  "ChildRemoved" ,  ""
2019-03-05 22:31:13 default: events:  "ChildRemoved" ,  ""
2019-03-05 22:31:13 default: events:  "PlatformSurface" ,  ""
2019-03-05 22:31:13 default: events:  "WinIdChange" ,  ""
2019-03-05 22:31:13 default: events:  "Destroy" ,  ""
Segmentation fault (core dumped)

I am wondering if it is a double delete, because handleDialogFinished delete's the dialog with in the event handler. I try changing delete _dialog to _dialog->deleteLater() and I have not seen the fault. not to say its not still there, am no expert when it come to QT.

guruofquality, 2019-03-05:
The delete later probably works because the object just happened to have some queued up events for the main qt thread to run on it, and the delete immediate is pulling the rug out from underneath of it, so to speak. What those events are, I'm not sure, but that's why it depends on the version of qt, timing of the close, or otherwise seems random.

Its probably not deleting twice, the dialog is stored in a QPointer, so the widget handler know if the pointer is deleted because delete also nulls it out.

So I think deletelater is safer, I just have to make sure that the QPointer is also cleared, since it uses it for the state of the widget as well. Thanks for testing that.

guruofquality, 2019-03-06:
I pushed the fix, hopefully its OK :_)

ashley-b, 2019-03-06:
Thanks, I will check it out and test it.

ashley-b, 2019-03-06:
Looks good. Thanks

closed 2019-03-06 by guruofquality