pothosware/PothosFlow #178: Error compiling 'moc_ConnectionPropertiesPanel.cpp'

gvanem, 2019-08-29:
I'm building for `x86` using MSVC-2019 (and clang-cl). But there is a problem with the generated stuff in `moc_ConnectionPropertiesPanel.cpp`. From clang-cl:
```cpp
...
In file included from ../../PropertiesPanel/ConnectionPropertiesPanel.hpp:6:
In file included from ../../GraphObjects/GraphEndpoint.hpp:6:
In file included from f:\Qt\5.12\5.11.2\msvc2015\include\QtCore\QPointer:1:
f:\Qt\5.12\5.11.2\msvc2015\include\QtCore/qpointer.h(86,14):  error: static_cast from 'QObject *' to 'GraphObject *', which are not related
      by inheritance, is not allowed
    { return static_cast<T*>( wp.data()); }
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~
../../GraphObjects/GraphEndpoint.hpp(96,52):  note: in instantiation of member function 'QPointer<GraphObject>::data' requested here
            (std::hash<size_t>()(size_t(s.getObj().data())) << 1);
                                                   ^
../../GraphObjects/GraphEndpoint.hpp(11,7):  note: 'GraphObject' is incomplete
class GraphObject;
      ^
1 error generated.
```

I love `clang-cl` for it's clear messages. Not something I can say for Cmake. 
With Cmake I tried with:
```
 cmake -G "NMake Makefiles" CMAKE_VERBOSE_MAKEFILE:BOOL=TRUE ..
 nmake -nologo -U
```

The `-U` was to print the inline-files Cmake is so found of, but I still failed to find a problem with the `CFLAGS`. Any hints?

Some assumptions on the size of pointers or a mismatch of `size_t`?

guruofquality, 2019-09-01:
The issue is that GraphObject is forward declared, and you must have a newer version of QPointer that static_casts its internal pointer. I only need the pointer for the hash, but it looks like I cant get it through the API any-more unless the type is defined.

Can you try this diff
```
diff --git a/GraphObjects/GraphEndpoint.hpp b/GraphObjects/GraphEndpoint.hpp
index 34f6f74..bf8ced8 100644
--- a/GraphObjects/GraphEndpoint.hpp
+++ b/GraphObjects/GraphEndpoint.hpp
@@ -3,13 +3,12 @@
 
 #pragma once
 #include <Pothos/Config.hpp>
+#include "GraphObjects/GraphObject.hpp"
 #include <QPointer>
 #include <QString>
 #include <QPointF>
 #include <QHash>
 
-class GraphObject;
-
 enum GraphConnectableDirection
 {
     GRAPH_CONN_INPUT,
```

PS not sure about the cmake flags, but if you want to see the build commands, usually with makefile its `make VERBOSE=1` or set the VERBOSE environment variable.

gvanem, 2019-09-02:
No, that made it fail more:
```cpp
In file included from PropertiesPanel/ConnectionPropertiesPanel.cpp:5:
In file included from PropertiesPanel/ConnectionPropertiesPanel.hpp:6:
In file included from .\GraphObjects/GraphEndpoint.hpp:11:
.\GraphObjects/GraphObject.hpp(71,25):  error: use of undeclared identifier 'GraphConnectableKey'
    virtual std::vector<GraphConnectableKey> getConnectableKeys(void) const;
                        ^
.\GraphObjects/GraphObject.hpp(72,13):  error: unknown type name 'GraphConnectableKey'
    virtual GraphConnectableKey isPointingToConnectable(const QPointF &pos) const;
            ^
.\GraphObjects/GraphObject.hpp(73,13):  error: unknown type name 'GraphConnectableAttrs'
    virtual GraphConnectableAttrs getConnectableAttrs(const GraphConnectableKey &key) const;
            ^
.\GraphObjects/GraphObject.hpp(73,61):  error: unknown type name 'GraphConnectableKey'
    virtual GraphConnectableAttrs getConnectableAttrs(const GraphConnectableKey &key) const;
                                                            ^
.\GraphObjects/GraphObject.hpp(77,19):  error: unknown type name 'GraphConnectableKey'
    virtual const GraphConnectableKey &currentTrackedConnectable(void) const;
                  ^
.\GraphObjects/GraphObject.hpp(127,31):  error: implicit instantiation of undefined template 'QPointer<QGraphicsObject>'
    QPointer<QGraphicsObject> obj;
                              ^
f:\Qt\5.12\5.11.2\msvc2015\include\QtCore/qmetatype.h(1349,1):  note: template is declared here
QT_FOR_EACH_AUTOMATIC_TEMPLATE_SMART_POINTER(QT_FORWARD_DECLARE_SHARED_POINTER_TYPES_ITER)
^
f:\Qt\5.12\5.11.2\msvc2015\include\QtCore/qmetatype.h(221,7):  note: expanded from macro 'QT_FOR_EACH_AUTOMATIC_TEMPLATE_SMART_POINTER'
    F(QPointer)
      ^
6 errors generated.
```

guruofquality, 2019-09-02:
My bad, its a circular include issue. I moved the definitions to a cpp source: https://github.com/pothosware/PothosFlow/tree/win_fixes

The most annoying part is that I think the hash is for std::unordered_map/set which I am not using.

closed 2020-01-22 by guruofquality

gvanem, 2020-02-28:
Hi again. The `win_fixes` tree is gone and the circular .hpp-issue is still there.
From `clang-cl`:
```c++
In file included from .\GraphObjects/GraphEndpoint.hpp:6:
.\GraphObjects/GraphObject.hpp(71,25): error: use of undeclared identifier 'GraphConnectableKey'
    virtual std::vector<GraphConnectableKey> getConnectableKeys(void) const;
                        ^
.\GraphObjects/GraphObject.hpp(72,13): error: unknown type name 'GraphConnectableKey'
    virtual GraphConnectableKey isPointingToConnectable(const QPointF &pos) const;
            ^
.\GraphObjects/GraphObject.hpp(73,13): error: unknown type name 'GraphConnectableAttrs'
    virtual GraphConnectableAttrs getConnectableAttrs(const GraphConnectableKey &key) const;
            ^
.\GraphObjects/GraphObject.hpp(73,61): error: unknown type name 'GraphConnectableKey'
    virtual GraphConnectableAttrs getConnectableAttrs(const GraphConnectableKey &key) const;
                                                            ^
.\GraphObjects/GraphObject.hpp(77,19): error: unknown type name 'GraphConnectableKey'
    virtual const GraphConnectableKey &currentTrackedConnectable(void) const;
                  ^
.\GraphObjects/GraphObject.hpp(127,31): error: implicit instantiation of undefined template 'QPointer<QGraphicsObject>'
    QPointer<QGraphicsObject> obj;
                              ^
```
But this seems to work:
```diff
--- a/GraphObjects/GraphEndpoint.hpp 2017-03-23 21:20:04
+++ b/GraphObjects/GraphEndpoint.hpp 2020-02-28 11:20:05
@@ -3,6 +3,7 @@

 #pragma once
 #include <Pothos/Config.hpp>
+// #include "GraphObjects/GraphObject.hpp"
 #include <QPointer>
 #include <QString>
 #include <QPointF>
```

guruofquality, 2020-02-29:
I merged win_fixes, that change should be in master