pothosware/PothosFlow #187: Missing Variables

dbnut, 2020-01-23:
Global variables, please?
Bind to input or output controls (including processing block properties like VCO.SampleRate).
Two-way (R/W) bindings as appropriate (e.g. slider).
Variable values to include arithmetic expressions (arguments to include other values and preferably direct reference to block properties).
Don't know how compatible that is with your model, but I couldn't use GRC without it.

guruofquality, 2020-01-23:
Yes there are global variables: https://github.com/pothosware/PothosFlow/wiki/Tutorial#global-variables

But its really just for setting properties at block creation time. It doesnt really fit the model, because the blocks are supposed to go be distributed to other processors and to give each other feedback through connections.

But I get a lot of requests to support the generate and execute style like GRC. I tried something different this time that people also wanted to see in GRC. I guess I need to have the best of both worlds and time.. (that too)

dbnut, 2020-01-23:
Global _constants_ :)

> and to give each other feedback through connections
Fine if they were more widely implemented, like updating a slider in response to a "senior" control (like a direct-entry frequency text box or odometer resetting several fine-tuning sliders). Have to be careful about signal loops, but i suppose the monitor would warn of that like other processing overloads.
I'll close this now, thanks, but will have a critical look at my GRC flows and see if your approach could work for me.

closed 2020-01-23 by dbnut