pothosware/PothosFlow #189: BlockEval doesn't distinguish between overlay() not being implemented and overlay() throwing an error

ncorgan, 2020-03-01:
When BlockEval calls a block's `overlay()` to get any custom updates for the block's display, it's done inside a try-catch that ignores the output on any error. However, the same behavior occurs if `overlay()` throws an exception or if it simply doesn't exist. In the case where the block throws an error inside `overlay()`, it would be helpful to log the error.

See here: https://github.com/pothosware/PothosFlow/blob/25b689b8c1790a4a89fb952206667a377920e98c/EvalEngine/BlockEval.cpp#L286

guruofquality, 2020-03-07:
Maybe i just need some good reflection support so the client class can ask if a block has an overlay. Or for that matter if any proxy object has a function, probably an addition to the proxy classes and their overloads.

ncorgan, 2020-03-07:
That would be useful and a better way to do it in the long term than my pull request. It makes it tedious to figure some issues out, so maybe pull it in with a TODO?

closed 2020-03-08 by guruofquality