pothosware/PothosFlow #191: PothosFlow will not load without active internet

Matthewenderle, 2020-03-18:
We're trying to utilize Pothos in an environment without internet connectivity. Upon launching the application it fails and gives an error. I've tried reinstalling, but I've been able to replicate the issue on another machine by disabling the network connection.

`Remote Server Error: Pothos::RemoteServer(tcp://[::1]:16415): Failed to launch server process: C:\Program Files\PothosSDR\bin\PothosUtil.exe --require-active --proxy-server=tcp://[::1]:16415`

Screenshot:
[](https://imgur.com/a/ZIOQqi4)

guruofquality, 2020-03-18:
Its not that it needs the internet, for some reason, it cant bind a socket to "::" (which is like the ipv6 equivalent of 0.0.0.0) when all network connections are down. Although its poco sockets, I think its really more of a windows issue. Its like how linux has the "lo" device, but the windows equivalent seems to go away without network connections. I wish I knew what to search for to just trick it or get the interface back.

guruofquality, 2020-03-21:
Just a thought, the gui will not try to spawn anything if it can find an existing process. Does running `PothosUtil --proxy-server="tcp://127.0.0.1"` in another terminal help?

closed 2020-03-21 by Matthewenderle

Matthewenderle, 2020-03-21:
I was able to get it figured out. I got a little busy and was unable to update you on this. My resolution came about when I realized that it was just looking for a network adapter to bind to. Since I wasn't going to get a good connection on my windows server I needed to create a loopback.

For the steps to run PothosUtil without a network connection on Windows:

- Open **Device Manager**.
- Select **Add Legacy Hardware** under the _Action_ tool bar.
- Click _Next_ and select _Install the hardware that I manually select from a list (Advanced)_.
- On the left pane choose _Microsoft_ and on the right find the loopback adapter. Mine was _Microsoft KM-TEST Loopback Adapter_ on Windows 10.
- Finish the process by clicking _Next_ until it's installed.
- Relaunch Pothos and it should now run.