pothosware/PothosFlow #192: Passing complex numbers into block slot from PothosFlow

ncorgan, 2020-03-22:
Unless I'm missing something, there doesn't seem to be a way to pass complex values into slots from PothosFlow. NumericEntry only work for scalar floats, and LineEdit returns a std::string, which can't convert to a complex number.

Is there an existing method/workaround to pass in complex values from a graph widget? If we need to use a workaround, there should be support in a graph widget.

guruofquality, 2020-03-22:
there is this: https://github.com/pothosware/PothosWidgets/wiki#planar-selection

I think two numeric entries and the eval block could be used as well to format the complex number from components

ncorgan, 2020-03-23:
Without showing the graph widget, I'd think it's an odd name to appear on a PothosFlow display when the user just wants a complex number.

It looks like it works though.